lnls-dig/rtm-lamp-hw #9: Driving the DAC output from high to low causes interference in the current measurement

Palmitoxico, 2021-06-17:
Channels 0, 2, 3, 5, 6, 8 and 9 are picking up interference from the DAC in the current measurement circuitry. Only when the DAC output swings from 4V to 0V the interference occurs suggesting it might be a grounding issue.

[image]

danielot, 2021-06-18:
Note: these measurements were taken when opamps were disabled/powered off.

Palmitoxico, 2021-06-21:
Removing R83 and soldering a 33 ohms resistor directly to the positive and negative inputs of the AD8429 (IC29) makes the glitch to from the input, but the output still behaving the same. One thing I observed it that the positive and negative power supplies, when measuring directly from the inamp pins, show a large voltage glitch (1 - 2Vpp) that might explain this issue:

[image]

Unfortunately, I couldn't reproduce this in spice, but maybe the [model](https://www.analog.com/media/en/simulation-models/spice-models/ad8429.cir) is not accurate enough.

[rtm-lamp-ad8429-simu.zip](https://github.com/lnls-dig/rtm-lamp-hw/files/6688372/rtm-lamp-ad8429-simu.zip)

danielot, 2021-06-21:
Do these glitches on the +/-7V power supply happen for positive DAC steps too? It would be nice to see the curves for the positive step as well.

Palmitoxico, 2021-07-02:
@danielot They do, but at a lower amplitude.

Palmitoxico, 2021-07-02:
OK now it seems I'm getting somewhere. Maybe my first attempt to solder an 33 ohms resistor directly into the AD8429 differential inputs wan't making good contact. I've re-soldered it and the glitch is almost gone in channel 8:

[image]

I'll try this for the other channels just to be sure.

Palmitoxico, 2021-07-08:
It is a grounding issue. After soldering wires to reduce the ground impedance for C85, C84, C88 and C89 the I was able to reduce the glitch amplitude significantly:
[image]

Fix 1:
[image]

Fix 2:
[image]

Palmitoxico, 2021-12-21:
Solved in the newer releases.

closed 2021-12-21 by Palmitoxico